tttapa/ESP8266 #2: Beginner's 8266 Guide - mDNS

jbltimberrock, 2017-04-16:
Pieter,
First, thank you very much for your detailed explanation of the ESP8266 module, its capabilities and functions. I have successfully worked through each 'tutorial up to and including the Turning On & Off an LED over WiFi. Everything worked perfectly EXCEPT the mDNS function. While the sketches verified and uploaded fine through my FTDI and ESP8266 ESP-01 and I received via serial that 'mDNS responder started' and 'HTTP Server Started' I could not receive a response when typing in http://esp8266.local. Ping to esp8266.local did not respond. HOWEVER, if I typed in the local IP, I could PING and receive "Hello world" and TOGGLE LED worked fine. Can you advise solution? Thank you in advance.

tttapa, 2017-04-18:
What OS are you using? Are your computer and the ESP on the same Wi-Fi network, or is there some network infrastructure in between, like a router? 
Does OTA work?
You could try to reboot your Wi-Fi router.

Pieter

jbltimberrock, 2017-04-19:
Pieter...thanks for response. I'm using Windows 7. Yes, my laptop and the 8266 are both connected to "linksys" wifi in the house w/o any other device in between. I reloaded it all today and I get the same results...Ping with the local IP address, Hello World and Toggle LED work fine that way, but again not when using esp8266.local. One other comment, my wifi does not use a password, so I commented or otherwise removed "password" from the code.
Tried Blink OTA...changed to pin2 for the led. Uploaded sketch by Serial...works fine. Then set Board Port to ESP8266@IP, Changed to 500ms and tried to upload OTA, but cannot get beyond the password prompt...won't accept it and keeps requesting password. John

tttapa, 2017-04-19:
I don't think W7 supports mDNS.
OTA won't work either, because it relies on mDNS.

Pieter

tttapa, 2017-04-19:
You can try installing iTunes, it includes Apple Bonjour to resolve mDNS addresses. You can extract it from the iTunes installer if you only want Bonjour and not iTunes. [https://superuser.com/questions/491747/how-can-i-resolve-local-addresses-in-windows](https://superuser.com/questions/491747/how-can-i-resolve-local-addresses-in-windows)  
Pieter

jbltimberrock, 2017-04-19:
Thanks Pieter. First I think I'll move over to my desktop that uses Windows 8.1 and see what happens. Don't know if this is relevant, but before my message yesterday and trying OTA, I had reflashed the AT firmware to the 8266. While commands such as AT, AT+RST, AT+GMR, AT+CWMODE, and AT+CWLAP worked fine, the command AT+CWJAP would not connect me to the available ssid indicated in CWLAP response. Also a W7 issue that Bonjour would resolve?

jbltimberrock, 2017-04-19:
Pieter...UPDATE...OK, set it all up on W8.1. connects to my wifi ssid 'linksys' OK and from CMD line, PING to IP works, but PING to esp8266.local does not work. John

tttapa, 2017-04-19:
Windows 8.1 doesn't support mDNS either, I think you need W10, or you have to install a third-party mDNS responder, like Apple's Bonjour. 
The AT+CWJAP command issue has nothing to do with mDNS, so installing Bonjour won't help with that.

By the way, I strongly suggest to put a password on your WiFi: a WiFi network without password is not encrypted. This means that anyone with a WiFi antenna can read all your network packets. If you have any network sharing service enabled (could be by accident, or by default) they could even access the files on your PC.

Pieter

jbltimberrock, 2017-04-19:
Pieter..first, thank you for your continued interest in helping me resolve these issues. OK, installed Bonjour and whaddya know, the esp8266.local worked for the 'toggle LED' as well as the 'Hello world' examples. However, I am not able to get beyond the password request on the OTA example. When I close the password box, I receive the error: "Error while uploading: missing 'runtime.pwd.192.168.1.102' configuration parameter." Same with GPIO0 grounded or ungrounded during this. 

Now, putting that aside for a moment, with Bonjour loaded, I went back and re-flashed the AT firmware and I find that AT+CWJAP works and does join my linksys wifi! wahoo! But, it does require a placeholder for a password...so, AT+CWJAP="linksys","" and this works

Lastly, for this post, thank you for your concern and the recommendation regarding wifi password security. I had considered that initially, but since we live on a large parcel of rural acreage where our closest neighbor is 1/2 mile distant I decided it was not necessary...our wifi is not detectable 100' outside the house.

My ultimate goal with the 8266 is to be able to detect something in our barn with a PIR and send a text or email using the IFTTT method

tttapa, 2017-04-20:
Glad to hear you got mDNS working!
Does the IDE see the network port under Tools > Port > Network Ports > ESP8266 at 192.168.xxx.xxx?
You don't have to ground GPIO0 during OTA updates.

I've never used IFTTT, so I can't help you with that.

Pieter

jbltimberrock, 2017-04-20:
Pieter,
OK, I tried Blink OTA again...same result...everything works as you write in the tutorial...I upload the sketch, reset the 8266 and get the blinking, close all IDE windows and reopen and select Tool>Port>ESP8266 at 192.168.1.102m, change from 1000 to 500 then start to upload again,,,as soon as compile finishes the password window comes up,but every time I enter the password esp8266, the password box blinks clear and again asks for a password, Here are the messages 'in orange' in the bottom pane of the IDE. I have Java 8 Update 45 installed, but I don't have any Python.exe installed that I am aware of...is that something that is being call inside ArduinoOTA.h ? John
java.io.IOException: Cannot run program "python.exe": CreateProcess error=2, The system cannot find the file specified at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.GenericNetworkUploader.uploadUsingPreferences(GenericNetworkUploader.java:99)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:709)
	at processing.app.SketchController.exportApplet(SketchController.java:682)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2190)
	at java.lang.Thread.run(Thread.java:745)

tttapa, 2017-04-22:
I think I manually installed Python on most of my Windows machines (for reasons other than Arduino), so maybe that's why it worked right out of the box for me.
I'll try it on a clean W10 VM to see if Python is required for OTA.

Pieter

jbltimberrock, 2017-04-22:
Thanks. BTW, I found this tutorial with a very clean code for sending an e-mail with the 8266...there is a Gsender.h library that removes a lot of the clutter from the .ino file. I modified it to work with a PIR sensor...just thought I'd pass it along:
http://tech.thejoestory.com/2016/10/esp8266-gmail-send-email.html

tttapa, 2017-05-06:
Thanks for the tip!

You need Python 2.7 for OTA to work, on most Linux systems this is installed by default, and I have it installed on most of my Windows PCs as well, so I didn't notice it.
I'll add it to the guide.

Thank you for your feedback,
Pieter

jbltimberrock, 2017-05-07:
Pieter,
Thank you for all your help. I'll look into the Python 2.7 and install it.
Thank you for spending a lot of time preparing a great tutorial. Cheers.
John

msameeruae, 2017-05-27:
Pieter, 
thanks for the awesome article on esp8266. I had the same problem with OTA thing and resolved it by getting some hint from arduino built in example. Now uploading my sketches wireless. I think the problem has something to do with "<WiFiUdp.h>" which your example code was not having.

bill-orange, 2017-09-24:
I have a little  info that may help others.

I had some frustration getting OTA to work.  It turns out my Time Warner / Spectrum router blocks multicast between wired and wireless.  I am updating my code wirelessly but only by switching my 'programming machine' out of wired and on to wireless when I am ready to program.  

The Time Warner / Spectrum router also apparently blocks  the' mDNS responder', hopelessly.  There does not seem to be a setting change or work-around that would make it work.  Fortunately, it is not essential.

hollandmike, 2017-09-28:
One of my machines runs W10, and couldn't connect using mDNSResponder (obviously).  I installed the Chocolatey package manager (https://chocolatey.org/), and then used that to install the Bonjour service (didn't need to mess with the rest of iTunes), using the command line "choco install bonjour".  Now it works fine.

proddy, 2017-10-28:
I had same issue with mDNS on a fresh win10x64 install with the latest iTunes. Followed @hollandmike suggestion and installed the bonjour service separately with 'choco install bonjour', so thanks for that.

closed 2018-01-02 by tttapa