tttapa/ESP8266 #3: Problem with Led Control example

Primus007, 2017-05-03:
Many thanks for your great work and the examples. They was very helpful for me.

I have a problem with the Led Control example. I use an ESP-01. I have uploaded the Data Dir and the sketch (i use the arduino ide 1.8.1 for this) . The Server starts:
Station connected to ESP8266 AP
OTA ready

Spiffs all ok 
WebSocket server started.
mDNS responder started: http://esp8266.local
HTTP server started.
handleFileRead: /
	Sent file: /index.html
handleFileRead: /main.css
	Sent file: /main.css
handleFileRead: /WebSocket.js
	Sent file: /WebSocket.js

!!! and here is the Problem. It need 1 Minute an 20 seconds to sent the file. Before and after this all runs fast. 

[0] Connected from 192.168.4.2 url: /
handleFileRead: /favicon.ico
	Sent file: /favicon.ico
[0] get Text: Connect Wed May 03 2017 15:21:11 GMT+0200 (CEST)
handleFileRead: /favicon-144x144.png
	Sent file: /favicon-144x144.png

Please can you help me to solve the problem?
many thanks.

Primus007, 2017-05-06:
I have found the solution:
The Problem is the WebSocket.js line number 5: `<link href='https://fonts.googleapis.com/css?family=Roboto:300' rel='stylesheet' type='text/css'>`
It tries to get the font over the web. but i use my esp as AP! Nothing can get from the web!
Now it works fine.

tttapa, 2017-05-06:
That's a bug indeed, I'll fix it when I find the time.
Thanks for your feedback.

Pieter

commit b38c80fe (2017-06-29): Fix #3
closed 2017-06-29 by tttapa