tttapa/ESP8266 #5: Data logging temperature

per-ast, 2017-10-22:
Hello

I like this A Beginner's Guide to the ESP8266.
Can not find description of how it works with the web page.
How is the website intended to work?
Should I run a local web server on my PC?
Should I upload the files to esp8266 (index.html)?

It looks like this

Connecting
.........

Connected to Eskilstorp
IP address:	192.168.15.48

OTA ready

SPIFFS started. Contents:
	FS File: /temp.csv, size: 2.41KB

mDNS responder started: http://esp8266.local
HTTP server started.
Starting UDP
Local port:	123
Time server IP:	132.163.96.2
Sending NTP request
NTP response:	1508680734
Temperature requested
Appending temperature to file: 1508680789,26.50
Temperature requested
Appending temperature to file: 1508680849,26.50
Temperature requested

Sincerely
P-G Åström

tttapa, 2017-10-22:
You have to upload the files to the ESP8266. It is explained in the _Using the example_ paragraph of [chapter 16](https://tttapa.github.io/ESP8266/Chap16%20-%20Data%20Logging.html):

> Set the SPIFFS size to 64KB or larger if you plan to use it for prolonged periods of time. (You could also increase the logging interval on line 80 to save space.)
Enter your Wi-Fi credentials on lines 138-140, and hit upload. Then upload the webpages and scripts to SPIFFS using Tools > ESP8266 Sketch Data Upload.
Make sure you have the temperature sensor connected, as described at the top of this page. Open a terminal to see if it works. [...]
Let it run for a couple of minutes, to gather some temperature data. Then open a web browser, and go to http://esp8266.local/.
You should get a graph showing the temperature curve. You can use the arrow buttons to travel through time, and the + and - buttons to zoom in or out. The reset button resets the zoom, and jumps back to the present. Refresh requests the latest temperature data.
If you want, you can still go to http://esp8266.local/edit.html to upload new files.

closed 2017-10-22 by tttapa

per-ast, 2017-10-22:
Hi

My problem is that I can not upload, for example, index.html. In the Tools meny I don’t have "ESP8266 Sketch Data Upload" Created the folder data with the files in

//P-G Åström

per-ast, 2017-10-22:
Hi 

It’s working.

How can I download the temp.csv file to my pc?

Sincerely
P-G Åström

tttapa, 2017-10-24:
Just use a browser, wget or cURL etc. to download http://esp8266.local/temp.csv.

per-ast, 2017-10-24:
Hello

Thanks. Perfect

Is it possible to connect several temp sensors?

Sincerely
P-G Åström