tttapa/ESP8266 #11: Auto-reset and auto-program on bare-bones ESP8266 using DTR only

Erik84750, 2017-12-14:
Besides using DTR to perform auto-reset (page 14) it is also possible to use DTR for auto-program (without the use of RTS): connect DTR to GPIO0 with a small signal diode (1N4148) in series, anode to GPIO0. Add a 10k pull-up between anode and 3.3V.
Tested on 12E and 12F. Schematics added for adapter board (which also doubles as adapter for RF95/96 modules) where this modification is included.
[ESP8266-12_auto-reset.pdf](https://github.com/tttapa/ESP8266/files/1557791/ESP8266-12_auto-reset.pdf)

Erik84750, 2017-12-16:
Hi tttapa, I need to come back on my comment above: I do experience issues that need further investigation. In some instances the ESP8266 does not revert to normal operating mode after upload of flash. I suspect it has to do with the use of the RESET in my setup. Please bear with me until I have this resolved and freeze this issue in the meantime. Thank you and sorry for inconveniences caused. Grts, Erik

Erik84750, 2017-12-24:
Hi tttapa, after a few days of testing I come with following conclusions for the auto-reset and auto-program circuit:
1. as mentioned above, connect DTR with diode (but use Schottky diode) to GPIO, anode to GPIO0
2. use your circuit, except: a. remove 680R resistor and b. connect to CH_PD
3. leave the RESET pin unconnected
[ESP8266-12_auto-reset_v2.pdf](https://github.com/tttapa/ESP8266/files/1584667/ESP8266-12_auto-reset_v2.pdf)
When connecting the FTDI USB-serial converter make sure either Arduino serial monitor is closed of remove the DTR connection to the ESP.
When programming connect the DTR to above contacts. The ESP will automatically enter programming (flash) mode when uploading, and after upload the ESP will automatically reset in normal operating mode. When the Arduino serial monitor needs to be connected remove the DTR connection beforehand.

Erik84750, 2018-01-02:
Hi tttapa, after several more days of testing and repeatability measuring I give up: there is currently no way I can guarantee a robust auto-reset and auto-program method. At the moment the only solid solution still is the one you describe in your manual. Until I find a solid and repeatable process and schematic I suggest to leave this subject froozen. Greetings and sorry for inconveniences caused, Erik

tttapa, 2018-01-02:
No problem, I appreciate you sharing your findings! :)

Erik84750, 2018-01-24:
Hi, after several weeks of testing I found a solution for the auto-reset: see attached schematic. If you send me a pm with your address I will send you a polulated pcb. The pcb can double as a adapter board for LORA RF95/96 radio modules.
[ESP8266-12_auto-reset_v3.pdf](https://github.com/tttapa/ESP8266/files/1660642/ESP8266-12_auto-reset_v3.pdf)

Erik84750, 2018-01-24:
Here is v4, corrected for swapped GPIO04 and GPIO05
[ESP8266-12_auto-reset_v4.pdf](https://github.com/tttapa/ESP8266/files/1660661/ESP8266-12_auto-reset_v4.pdf)

Erik84750, 2018-01-24:
Here is v4, corrected for swapped GPIO04 and GPIO05
[Uploading ESP8266-12_auto-reset_v4.pdf…]()

Erik84750, 2018-01-24:
Two caveats for ESP8266 programming:
1. all GPIO's have to be disconnected except those needed for programming (per schematic)
2. restart completely (disconnect from any wire connections) to start the program

Trance-Paradox, 2020-05-07:
Does it works when you open serial monitor and try to send some serial data to it?

Erik84750, 2020-05-07:
"Does it works when you open serial monitor and try to send some serial data to it?" I am not sure I understand this question. The purpose of my setup is to be able to program the ESP8266 with an automatic reset after programming the ESP. Why would anyone want to use the serial monitor while downloading a program to the ESP8266?

Trance-Paradox, 2020-05-07:
> "Does it works when you open serial monitor and try to send some serial data to it?" I am not sure I understand this question. The purpose of my setup is to be able to program the ESP8266 with an [...]

No no I am not asking about using a serial monitor while downloading i am asking of opening up serial monitor or sending some serial commands resets the esp?
Can we use the serial monitor with the esp while not downloading anything just to debug?

Erik84750, 2020-05-08:
Yes you can. But your question is off-topic here.

Trance-Paradox, 2020-05-08:
> Yes you can. But your question is off-topic here.

Thank you I will try it and report back

Erik84750, 2020-05-08:
You do not understand: your issue does not belong here. Please open a new topic in a relevant forum.

Trance-Paradox, 2020-05-08:
> You do not understand: your issue does not belong here. Please open a new topic in a relevant forum.

Okk I understand

Trance-Paradox, 2020-05-09:
The reset circuit was working but for some reason my GPIO0 was not being pulled down.  So I used a pnp transistor instead of a diode to pull it down to GND and everything works fine. Also this curcuit will not enter download mode unless the pull down time of DTR pin is greater than reset time of the chip. Which is great for me. 
Thank you bro this post was  a big help.