tttapa/ESP8266 #13: use of deleted function 'ESP8266WebServer

seamiki, 2018-01-27:
just a small update for the WebSocket example
ESP8266/Examples/14. WebSocket/A-WebSocket_LED_control/A-WebSocket_LED_control.ino
fails to compile showing following errror in the arduino ide:

> use of deleted function 'ESP8266WebServer::ESP8266WebServer(const ESP8266WebServer&)'

To fix, replace line 11
`ESP8266WebServer server = ESP8266WebServer(80); `
with
`ESP8266WebServer server (80);`
Thanks for the excellent tutorial

tttapa, 2018-03-26:
This is because the copy constructor was deleted in release v2.4.0 of the Arduino Core. The solution you proposed is indeed correct.

closed 2018-03-26 by tttapa

facetime88, 2019-01-22:
How do I can declare it within a custom class.
ESP8266WebServer server(80); will not working.

dashiva91, 2019-01-24:
> How do I can declare it within a custom class.
> ESP8266WebServer server(80); will not working.

Could you tell what are the issues?
If you declare it as a class member, you need to initialize port number with initialization list instead of constructor.

b4dnewz, 2019-02-20:
it's possible to initialize the server as private (or public) property of a class?

```c++
class SetupServer {
  ESP8266WebServer server(80);

  public:
    SetupServer(int port = 80) {
      // init server using custom port
    };
};
```
I'm not very familiar with c++ and i've tried different ways without success, any hint?

thank in advance.

mszekiel, 2019-04-25:
> I'm not very familiar with c++ and i've tried different ways without success, any hint?
> 
> thank in advance.

Hi. Did you solve this? I'm actually in quite similar point.

b4dnewz, 2019-04-26:
@mszekiel nope.. 

I think the solution is in @dashiva91 answer:
> If you declare it as a class member, you need to initialize port number with initialization list instead of constructor.

but I'm not sure how to implement it

tttapa, 2019-04-26:
You cannot initialize the member like that. It's just a declaration, not a call to the constructor. You can provide a default value, though, but you cannot use [direct initialization](https://en.cppreference.com/w/cpp/language/direct_initialization).  
The default value will be used to initialize the member in the constructor when no other initialization is specified. This is the important bit: initialization happens in the constructor, not at the declaration of the member.

This will not work:
```cpp
class SetupServer {
  private:
    ESP8266WebServer server(80); // <-- Trying to call constructor ESP8266WebServer::ESP8266WebServer(int)
};
```

This is better:
```cpp
class SetupServer {
  public:
    SetupServer() = default; // <-- `server` is implicitly initialized to ESP8266WebServer{80} here
    SetupServer(int port) : server{port} {} // <-- Default value is ignored here
    
  private:
    ESP8266WebServer server = {80};  // <-- Declaration with default value using list initialization
};
```

Note the use of an [initializer list](https://en.cppreference.com/w/cpp/language/initializer_list) to initialize `server` in `SetupServer::SetupServer(int)`.

In this case, you could just leave out the default value entirely, because there is only a single constructor that uses it.
```cpp
class SetupServer {
  public:
    SetupServer(int port = 80) : server{port} {}
    
  private:
    ESP8266WebServer server;
};
```

If you had different constructors that needed the default value, the previous snippet would be bad, because you would have to repeat yourself:
```cpp
class SetupServer {
  public:
    SetupServer(int port = 80) : server{port} {}
    SetupServer(string name) : server{80}, name{name} {} // <-- repetition of default value
    
  private:
    ESP8266WebServer server;
   string name;
};
```
This would be better:
```cpp
class SetupServer {
  public:
    SetupServer() = default; // <-- `server` is implicitly initialized to ESP8266WebServer{80} here
    SetupServer(int port) : server{port} {} // <-- Default value is ignored here, name is default initialized to string{}
    SetupServer(string name) : name{name} {} // <-- `server` is implicitly initialized to ESP8266WebServer{80} here as well
    
  private:
    ESP8266WebServer server = {80};  // <-- Declaration with default value using list initialization
    string name;
};
```
Some more links:  
- [Member initialization](https://en.cppreference.com/w/cpp/language/data_members#Member_initialization)
- [List initialization](https://en.cppreference.com/w/cpp/language/list_initialization)
- [Initializer list](https://en.cppreference.com/w/cpp/language/initializer_list)

b4dnewz, 2019-04-26:
@tttapa dude that's great! this evening i'm going to study some of these links
__many__ __many__ thanks

tttapa, 2019-04-26:
The links to cppreference contain a lot of information, don't try to understand everything on your first read. Especially the special cases and the different behaviour between C++ versions (the ESP8266 Core uses C++11 - don't ask me why).  
Try to focus on the explanation and the examples.