tttapa/ESP8266 #15: It can be used in both Slave and Master mode (in software).

svofski, 2018-02-24:
I think this phrase is very vague. What does "(in software)" mean here? Does it mean that SPI is bit-banged (the paragraph about I2C above used "in software" in this meaning but is clear there), or is only one of the two modes bit-banged, or does it just refer to the fact that SPI is exposed to the user? 

Fantastic guide BTW, thank you so much for putting it together!

tttapa, 2018-02-24:
The early versions of the datasheet talk about HSPI as an SPI slave only, saying "SPI mode can be implemented via software programming." Newer versions say it's "Slave/Master". Afaik, it's probably mostly hardware, so I'll just leave out the "in software" in the next revision.  
Thank you for the feedback!

svofski, 2018-02-24:
Is there a datasheet we can reliably refer to, or does it come from studying SDK or UDK? I'm just 2 days into esp8266 world and finding the information seems to be a bit chaotic task, lots of slightly contradictory bits all over the place.

From what I can see in the Arduino core source, the SPI master is implemented in hardware.