tttapa/ESP8266 #17: Thank you very much!

DocMors, 2018-03-28:
I've read an unknown number of tutorials over the years but this one told what I wanted to know without me having to squeeze. Great work, should you decide to write more tutorials I'll surely will read them.

tunx404, 2018-03-29:
Me too, thank you so much! I'm new to ESP8266 and your tutorial is very detailed and easy to understand. Keep up the good work!

uggima, 2018-04-07:
Had to leave a comment to say thanks, not all that new to coding for the ESP8266 but still the data logging + google graphs was a bit of an inspiration to me.

ttump, 2018-06-06:
Many thanks for an excellent article, certainly one of the best I've read. You cover all the important topics for using the 8266 on the web very clearly and concisely. Great work!