tttapa/ESP8266 #22: websockets example not working

perigalacticon, 2018-05-18:
I am trying the websockets example for the first time.  I saved all 9 files in the data directory to SPIFFS.  When I open http://esp8266.local or the IP listed in the serial monitor I get in Chrome:

`This site can’t be reached
The webpage at http://192.168.0.10/ might be temporarily down or it may have moved permanently to a new web address.
ERR_CONTENT_DECODING_FAILED`

On the serial monitor I get:

`Opening port
Port open

Access Point "WOW!3630" started

Connecting
.........

Connected to WOW!3630
IP address:	192.168.0.10

OTA ready

SPIFFS started. Contents:
	FS File: /apple-touch-icon-180x180.png, size: 45.39KB
	FS File: /edit.html.gz, size: 45.17KB
	FS File: /favicon-144x144.png, size: 45.29KB
	FS File: /favicon.ico.gz, size: 45.19KB
	FS File: /index.html.gz, size: 45.18KB
	FS File: /main.css.gz, size: 45.04KB
	FS File: /manifest.json, size: 50.78KB
	FS File: /success.html.gz, size: 45.20KB
	FS File: /WebSocket.js.gz, size: 45.09KB

WebSocket server started.
mDNS responder started: http://esp8266.local
HTTP server started.
handleFileRead: /
	Sent file: /index.html.gz`

and a similar response for /edit.html.

Can you give me some suggestions to get it to work?  BTW I had to change lines 11-12 in A-WebSocket_LED_control.ino for it to compile without errors with VisualMicro IDE.  I changed them to:

`ESP8266WebServer server(80);       // create a web server on port 80
WebSocketsServer webSocket(81);    // create a websocket server on port 81
`
instead of:

`ESP8266WebServer server = ESP8266WebServer(80);       // create a web server on port 80
WebSocketsServer webSocket = WebSocketsServer(81);    // create a websocket server on port 81
`

eyer1951, 2018-08-05:
I had the same problem. Chrome doesn't like receiving gzipped index file. Researching how the gz files get unzipped before being served by ESP2866.

eyer1951, 2018-08-06:
OK, here's what solved this for me:  I had (wrongly) assumed that i could move each of the files in the data directory (here: [https://github.com/tttapa/ESP8266/tree/master/Examples/14.%20WebSocket/A-WebSocket_LED_control/data] ) to my project's data directory by right-clicking each and using "Save link as...". That doesn't work. The links are to a different page (one page for each of the files). Go to each of those pages in turn, and download, or right-click and save page as, or whatever it takes to get only the file itself to download. It's OK if the file is gzipped. Being a newbie (with github), there's surely a better way to transfer the contents of this github data directory to my local machine. If anybody would care to give that hint, I'd appreciate it.

Nickrader, 2018-08-07:
If you go to the root https://github.com/tttapa/ESP8266, you can clone/download the whole enchilada.