tttapa/ESP8266 #25: A Beginner's Guide to the ESP8266 - NTP part

DMike92, 2018-06-10:
Hi,
Thank you very much for this so complete tutorial that deals with lot of useful concepts!
I fully implement this and have a question about NTP.
Somewhere in the code you write:
`if (time) {                                  // If a new timestamp has been received
    timeUNIX = time;
    Serial.print("NTP response:\t");
    Serial.println(timeUNIX);
    lastNTPResponse = currentMillis; // <============
  } else if ((currentMillis - lastNTPResponse) > 3600000) {
    Serial.println("More than 1 hour since last NTP response. Rebooting.");
    Serial.flush();
    ESP.reset();
  }

  uint32_t actualTime = timeUNIX + (currentMillis - lastNTPResponse)/1000; // <========
  if (actualTime != prevActualTime && timeUNIX != 0) { // If a second has passed since last print
    prevActualTime = actualTime;
    Serial.printf("\rUTC time:\t%d:%d:%d   ", getHours(actualTime), getMinutes(actualTime), getSeconds(actualTime));
`
But lastNTPResponse = currentMillis; clearly means that currentMillis - lastNTPResponse will always be 0 when computing actualTime.
So, I'm a bit lost. What was your idea?

tttapa, 2018-06-10:
```cpp
unsigned long intervalNTP = 60000; // Request NTP time every minute
unsigned long prevNTP = 0;
unsigned long lastNTPResponse = millis();
uint32_t timeUNIX = 0;

unsigned long prevActualTime = 0;

void loop() {
  unsigned long currentMillis = millis();

  if (currentMillis - prevNTP > intervalNTP) { // If a minute has passed since last NTP request
    prevNTP = currentMillis;
    Serial.println("\r\nSending NTP request ...");
    sendNTPpacket(timeServerIP);               // Send an NTP request
  }

  uint32_t time = getTime();                   // Check if an NTP response has arrived and get the (UNIX) time
  if (time) {                                  // If a new timestamp has been received
    timeUNIX = time;
    Serial.print("NTP response:\t");
    Serial.println(timeUNIX);
    lastNTPResponse = currentMillis;
  } else if ((currentMillis - lastNTPResponse) > 3600000) {
    Serial.println("More than 1 hour since last NTP response. Rebooting.");
    Serial.flush();
    ESP.reset();
  }

  uint32_t actualTime = timeUNIX + (currentMillis - lastNTPResponse)/1000;
  if (actualTime != prevActualTime && timeUNIX != 0) { // If a second has passed since last print
    prevActualTime = actualTime;
    Serial.printf("\rUTC time:\t%d:%d:%d   ", getHours(actualTime), getMinutes(actualTime), getSeconds(actualTime));
  }  
}
```
`lastNTPResponse` is _only_ set to `currentMillis` if there is a response from the NTP server (i.e. if `time` != 0). `currentMillis` is updated every time the loop updates.

So when there are no responses from the NTP server, `lastNTPResponse` is never updated, but `currentMillis` keeps on growing.

DMike92, 2018-06-10:
I see, Thanks :-)
(Sorry for mistyping and closing)

DMike92, 2021-02-23:
Better later than never...

closed 2021-02-23 by DMike92