tttapa/ESP8266 #34: Will this work on the Adafruit HUZZAH32 - ESP32 board?

tpitman, 2018-10-22:
Will this work on the Adafruit HUZZAH32 - ESP32 board?

ilioss, 2020-06-16:
Hello tpitman,

I alsop hat the question if this also worked on a ESP32.
Not out of the box. So I went puzzling to have finally a working script on the ESP32. 
Note with the arduino ide it works on the PlatformIO it remains resetting the ESP32.
Hoppefully it is of help for you.
Regards,
ilioSS
`/*   ESP32-Async
 *
 *  dd 10-05-2020   File:  ESP32-Async_01
 *                         A-Temp. van tttapa van esp8266 naar esp32 proberen om te zetten?????
 *  dd 15-06-2020   File:  ESP32-tttapa-temp.logger_01
 *                         De esp8266 file omgebouwd naar de ESP32 lijkt te werken???????
 *  dd 16-06-2020   File:  ESP32-tttapa-temp-logger_04
 *                         Werkt en opgeschoond
 * 
 * 
 * 
 */
 #include <Arduino.h>
 
//**************************** NETWORK ****************************************** 
 #include <HTTP_Method.h>
 #include <WebServer.h>
 #include <WiFi.h>
 #include <WiFiUdp.h>
 #include <WiFiMulti.h>
 #include <ESPAsyncWebServer.h>
 #include <ESPmDNS.h>
 #include <AsyncTCP.h>
 #include <AsyncWebSocket.h>
 #include <ArduinoOTA.h>
 
 //************************* DS18B20 ********************************************
 #include <OneWire.h>
 #include <DallasTemperature.h>
 #define TEMP_SENSOR_PIN 5                                    // PAS OP DIT IS OP PIN  D2!!!!!!!!!!!!!!!!!!!!!!!!
 OneWire oneWire(TEMP_SENSOR_PIN);                              // Set up a OneWire instance to communicate with OneWire devices
 DallasTemperature tempSensors(&oneWire);                      // Create an instance of the temperature sensor class

 //********************* SPIFFS ************************************************
 #include <FS.h>
 #include <SPIFFS.h>
 #define FORMAT_SPIFFS_IF_FAILED true
 
 #define ONE_HOUR 3600000UL                                    // timer for request time
  
 WebServer server(80);

 File fsUploadFile;

 WiFiMulti wifiMulti;

 WiFiUDP UDP;

IPAddress timeServerIP;                                       // The time.nist.gov NTP server's IP address
const char *ntpServerName = "time.nist.gov";
const int NTP_PACKET_SIZE = 48;                               // NTP time stamp is in the first 48 bytes of the message
byte packetBuffer[NTP_PACKET_SIZE];                           // A buffer to hold incoming and outgoing packets

const char *OTAName = "ESP32";                                // A name and a password for the OTA service
const char *OTAPassword = "esp32";
const char *mdnsName = "esp32";                               // Domain name for the mDNS responder

const unsigned long intervalNTP = ONE_HOUR;                   // Update the time every hour
unsigned long prevNTP = 0;
unsigned long lastNTPResponse = millis();

const unsigned long intervalTemp = 60000;                    // Do a temperature measurement every minute
[... 274 lines cut ...]
      Serial.println("Temperature requested");
    }
    if (currentMillis - prevTemp > DS_delay && tmpRequested) {  // 750 ms after requesting the temperature
      long unsigned int actualTime = timeUNIX + (currentMillis - lastNTPResponse) / 1000;
                                                          // The actual time is the last NTP time plus the time that has elapsed since the last NTP response
      tmpRequested = false;
      float temp = tempSensors.getTempCByIndex(0);        // Get the temperature from the sensor
      temp = round(temp * 100.0) / 100.0;                 // round temperature to 2 digits

      Serial.printf("Appending temperature to file: %lu,", actualTime);
      Serial.println(temp);
      File tempLog = SPIFFS.open("/temp.csv", "a");       // Write the time and the temperature to the csv file
      tempLog.print(actualTime);
      tempLog.print(',');
      tempLog.println(temp);
      tempLog.close();
    }
  } else {                                                // If we didn't receive an NTP response yet, send another request
    sendNTPpacket(timeServerIP);
    delay(500);
  }
  server.handleClient();                                 // run the server
  ArduinoOTA.handle();                                   // listen for OTA events
 }
 `

tpitman, 2020-06-16:
I am not sure what you are asking. The plugin will have nothing to do with
resetting the esp32.

I do have several projects of my own that use the esp32. I have not used
the esp8266, but it should work fine.

If the esp32 is resetting then you have a bug in your code for it. I can't
tell just from looking at the code.

What I usually do in a case like this is put a bunch of
Serial.println("1");, then Serial.println("2");, etc through the code and
then watch the serial output to see how far it gets. That will usually then
point close to where the problem is. Then I can add more serial output
around where the problem is to track down what is wrong.

That is all I can suggest.

ilioss, 2020-06-16:
Hi tpitman,Thanks for your measage.I do not have a question.After my succes to adapt the esp8266 script to esp32 i wants jou just send the file. While i understand you asked if it works for esp32. Thats all.Hope it works for you. RegardsilioSSVerzonden vanaf mijn Samsung-apparaat

"@type": "EmailMessage",

"@type": "ViewAction",

"target": "https://github.com/tttapa/ESP8266/issues/34#issuecomment-644758114",

"url": "https://github.com/tttapa/ESP8266/issues/34#issuecomment-644758114",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"@type": "Organization",

"name": "GitHub",

"url": "https://github.com"

}

}

]

tpitman, 2020-06-16:
Oh, awesome. Thank you so much. I get so many requests for help I wasn't
sure.