tttapa/ESP8266 #36: Excellent work

Joaquininsky, 2018-11-04:
Excellent work, Pieter. It is an article very well explained, very complete and with clear examples. An excellent help for those of us who are starting in the ESP8266. Thank you for your effort..

jcldio, 2018-11-16:
I agree. Thank you for the amazing guide!

Craig1516, 2018-11-18:
Yep, I have to pile on to this parade of gratitude.  This is a wonderful body of work and has been very helpful to my understanding.  It only took me two days to read it the first time.  The next time, will take me much more time to focus on the nuances.  Thank you!

karandikarsm, 2018-12-04:
You have done excellent work - very systematic and great flow because you seem to know intuitively how to put it so the reader will understand easily.Hats off to you!
Any advice for Alexa Endpoint developers?

NickyNoo1963, 2019-01-14:
Absolutely brilliant! A well-written guide that superbly explains a complex subject in a logical manner. 
You should definitely consider writing a technical book!

ptrooms, 2019-11-04:
Thumbs Up !!!  Great and wonderful, very well written, staged  while getting knowledge and explained.