tttapa/ESP8266 #44: Broken ESP afterwards running example 14-

HamzaHajeir, 2018-12-26:
Hello tttapa, 

I've uploaded both sketch and data of Websocket example to ESP8266 nodeMCU .

It worked well for about 3 hours, Then something inside the program crashes and ESP keeps restarting twice a second! 

What makes it terrible that i can't even upload another code to it!

Can you provide a solution please ?

Thanks
Hamza Hajeir

lorol, 2019-04-12:
Hamza, you can start by verifying the power supply voltages at 5 and 3V3 lines . nodeMCU 3.3V may be "cranking" and device restarts.

HamzaHajeir, 2019-04-15:
Hi @lorol 
Well, I've disassembled ESP-12E and soldered appropriate pogo pins to have an ESP-12F programmer (and that worked xD).

so, I don't think it's a power issue. indeed the 'broken' esp-12e could run normal sketches.

Now I've uploaded a sketch into it: Examples -> ESP8266WebServer -> HelloServer.ino
core : 2.4.2

and analyzed the stack results using [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder), got the following 👍 

```
Decoding stack results
0x4010020c: _umm_free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1295
0x401006dc: free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1755
0x40106475: __wrap_spi_flash_read at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_phy.c line 267
0x401068a0: vPortFree at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\heap.c line 59
0x40106475: __wrap_spi_flash_read at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_phy.c line 267
0x402032db: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 156
0x402031f7: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 104
0x4010020c: _umm_free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1295
0x401006dc: free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1755
0x401068a0: vPortFree at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\heap.c line 59
0x40207e98: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char) at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\HardwareSerial.cpp line 50
0x402029aa: setup() at C:\Users\Hamza\AppData\Local\Temp\arduino_modified_sketch_308206/HelloServer.ino line 42
0x40208da4: loop_wrapper() at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp line 122
0x4010020c: _umm_free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1295
0x401006dc: free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1755
0x40106475: __wrap_spi_flash_read at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_phy.c line 267
0x401068a0: vPortFree at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\heap.c line 59
0x40106475: __wrap_spi_flash_read at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_phy.c line 267
0x402032db: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 156
0x402031f7: ESP8266WiFiSTAClass::begin(char const*, char const*, int, unsigned char const*, bool) at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 104
0x4010020c: _umm_free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1295
0x401006dc: free at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc\umm_malloc.c line 1755
0x401068a0: vPortFree at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\heap.c line 59
0x40207e98: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char) at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\HardwareSerial.cpp line 50
0x402029aa: setup() at C:\Users\Hamza\AppData\Local\Temp\arduino_modified_sketch_308206/HelloServer.ino line 42
0x40208da4: loop_wrapper() at C:\Users\Hamza\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp line 122
```

I guess it's a defected flash memory, Not sure yet.