tttapa/ESP8266 #51: Possible creating multiple charts?

hachler, 2019-04-02:
I loved too much this data logging but I would need to have around 3 different charts on the screen for my purpose project. Is it hard to add more charts in these current project? I am trying to play inside the .js files but not having lucky.

Thanks

m19858484, 2019-04-26:
Hi,
I have the same questions.
I create the multi CSV file, and I add multi-chart on .html, but I have no idea of how to edit the .js file.

tttapa, 2019-04-26:
I don't have much time on my hands right now, and my JavaScript is quite rough, so I can't really help you.

I think the Google Chart documentation is a good place to start. Loading the CSV files from the ESP8266 is trivial, getting the data to display on the graph is harder, because of the callbacks.

You could try adding another chart in the [drawChart](https://github.com/tttapa/ESP8266/blob/6b54006dc93ca627050012008cc102267f03b8c8/Examples/16.%20Data%20logging/A-Temperature_logger/Uncompressed_data/temperatureGraph.js#L66) function with the same data and options first, then duplicate the CSV loading part to display the right data, and use different options for your second chart.

Note that this is primarily a JavaScript question, so you might get more help on sites like StackOverflow.

Good luck!

m19858484, 2019-05-04:
I set different functions name in .js.
Finally, it works!
Thanks,

tttapa, 2019-05-04:
Glad to hear!

It might be useful to others if you could post your solution.

netless-ww, 2021-02-16:
I'm looking at the same problem. 
Any chance you could post your multi chart HTML & Jarva code M19858484? 
Would be much appreciated.

m19858484, 2021-02-16:
Hi netless-ww,
I've placed the src code here:
[code.zip](https://github.com/tttapa/ESP8266/files/5988899/code.zip)
Hope this helps!

netless-ww, 2021-02-18:
Hi 19858484
Just what I needed.
Thanks!

netless-ww, 2021-04-05:
Hi all

Thanks again to M19858484 for his input.

Here is my attempt at a more direct extension of Pieter P's Web Based Temperature Logging Program to Cater for two data streams.

Extension to more data streams should now be straightforward. 

Hope this helps others needing to log more than 1 data stream.

[ESP8266WebDatalogger.zip](https://github.com/tttapa/ESP8266/files/6257703/ESP8266WebDatalogger.zip)