tttapa/ESP8266 #57: Needed to add a service before I could get mDNS working

MProx, 2019-05-11:
Hi, 

At first I couldn't get the mDNS working on my NodeMCU, but I eventually managed to get it running by adding the following line to the end of the setup() function:

MDNS.addService("http", "tcp", 80);

This line is present in the mDNS example sketch, which is how I worked out that I needed it.

Thanks,
M

aramperez, 2019-05-12:
Out of curiosity, is mDNS working in STA mode, AP mode or both?

Thanks,
Aram

MProx, 2019-05-12:
I'm using it as a station on my home wifi network. I used the exact code from tttapa's tutorial, modifying only the SSID and password. The only way I could get it to work was to invoke MDNS.update() right at the beginning of loop().

aramperez, 2019-05-12:
I can't get mDNS to work in AP mode :-(.

Thanks!

MProx, 2019-05-12:
Hmm, does it respond to a ping (either using `esp8266.local` or the P address)? 

I also found that `MDNS.update();` has to appear before `server.handleClient();` in the `loop` function.

aramperez, 2019-05-13:
Ok, I have my MDNS.update after server.handleClient. I'll give that a try in the new few days, I'm busy with other stuff right now.

Thanks!

closed 2024-08-30 by MProx