tttapa/ESP8266 #60: Thank You and Question

Buzzardbait, 2019-06-30:
Thank you so much for a great introduction to the ESP8266. I have this nagging question though. I'm bewildered by memory management in the ESP8266. I have been experimenting with one of those cheap shields with 4 dip switches and the reset button. After fixing the J2Y FET problem replacing them with the correct J1Y FET it seems to be pretty much functional. So I burned the AI-Thinker AT binary to to flash at address 0x0 and AT commands work. I can even connect to my WiFi and it gets assigned an IP address as a station. All this was done via an FT232R USB device. Here's where it goes downhill. If I try to load a sketch using the IDE it zaps the AT capability and it doesn't work. It also zaps the AT support that I had downloaded so I had to reflash. How do you control the sketch destination and isn't there a volatile RAM somewhere where a sketch resides? I havn't seen anything talked about but flash memory which on my shield happens to be 4MB.

tttapa, 2019-06-30:
I'm not sure I understand your question. 

The AT firmware is just a program that you upload like any other program. The same is true for programs you write using Arduino. You cannot execute two binaries simultaneously.  
Why would you limit yourself to AT commands if you can write your own program using the Arduino IDE?

Why do you want to upload a sketch to volatile RAM?  
Sketches are uploaded to the flash memory chip, and the ESP boots from there.

Buzzardbait, 2019-06-30:
Well, according to the IDE your simple one button WebServer/LED sketch was about 1MB of "program memory" (Flash, I presume) and 27552 bytes for "program variables" out of a maximum possible (according to the IDE) of 81920 bytes. That's what I'm referring to when I talk about volatile Ram. So the IDE reports it when you compile but nobody ever talks about it when giving specs about the boards/chips. Anyway, after uploading the code it just sits there and doesn't report any IP address as having been assigned and neither the STAtion or the Access Point LEDs illuminate. PC shows nothing in network representing a another station has been added. So either the chip is bad(doubtful) or the download didn't work. The only message I got from the IDE was "Finished - by RTS" but there is no RTS/CTS handshaking going on with the debug/upload pins only providing Rx and Tx and they are properly reversed. So this cheap board is a mystery right now and I ordered a SparkFun Feather Hoopsie or whatever so I can quit wasting time on this thing. All the bells & whistles settings for these clone boards (like ck and Expressif Firmware choices) are no help to get something running. Thanks for your help though and the good overview.