tttapa/ESP8266 #61: The GPIO pins are 5V tolerant

Bartvelp, 2019-06-30:
Hi,

In your hardware guide you indicate the io pins are not 5v tolerant, while they are.
Source: [ba0sh1.com](https://www.ba0sh1.com/blog/2016/08/03/is-esp8266-io-really-5v-tolerant/
)

gmacario, 2019-07-04:
,,

Tech-TX, 2019-09-01:
Just a bit of follow-up documentation: here's the article from Hackaday where they opened the question of 5V tolerance of the GPIOs, and the CEO of Espressif responded with a Facebook update that said yes, the GPIO pins are 5V tolerant:
https://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/#comment-3122361
The Facebook comment from Teo Swee Ann was “I can reply officially here: it is 5V tolerant at the IO. while the supply voltage is at 3.3V.”

The test done by ba0sh1 (above) is also linked in the Hackaday article.

The module power is NOT 5V tolerant, as the flash chip may die with 5V on it's power pin.  That suggests that GPIO6-11 on ESP-12 modules and DevKits (SC, SK, S0-S3) are probably also not 5V tolerant as they connect directly to the flash chip.  The RST pin on the module is not 5V tolerant, either.

Mohammad271091, 2019-09-15:
can you guys plz answer my issue #66 about GET requests ...

Buzzardbait, 2019-10-18:
In general, I always use level translators on the I/O pins of the NodeMCU SPI except for one pin. You cannot use a level translator on the SPI SS (slave select pin) because it will lock up the NodeMCU). I've also used the I2C pins without level translators and they work ok. Or at least they havn't blown up yet. :-)

mk-pmb, 2020-02-26:
I'm all for at least mentioning the CEO claim, albeit with a warning that other technical parameters might deviate from specs when used with out-of-spec inputs, as mentioned in the Hackaday comments.

Also a preliminary warning: When asked about the opening post's article, one of my advisors on electronics said the experiment seems to have some serious flaws we'll need to discuss, so you might want to ask your own advisor before relying on that report. (I'll update this warning as soon as I understand the concerns.)