tttapa/ESP8266 #66: GET Requests

Mohammad271091, 2019-09-14:
Hello, 
I'd like to say this is the AWESOMEST tutorial I've seen about ESP boards ! Original work .. 

My question: why do we send a GET request when we want to send our user and password to Gmail server ? why POST doesn't work in this case ? as you told, the GET request sends the form controls into the URI which is not safe at all in this case, encoding is not enough when it comes to passwords, no ? 

The second issue is that the email notifier didn't work for me, it failed to connect to google server, is there an update or I've missed something ? .. 

I hope I find an answer..
Thanks in advance,

tttapa, 2019-09-17:
There is nothing inherently unsafe about encoding the credentials in the URI of an HTTP GET request. The format in the body of a POST request would be exactly the same, and is not encrypted (not by HTTP at least). The reason that it's best not to use credentials in a GET request is that it is saved in the cache or history of web browsers. That's not the case for the ESP8266.

All HTTPS data is encrypted using TLS, so your password should be safe.

The example is rather old, I wrote it 3 years ago, so the Google API has probably changed.

Mohammad271091, 2019-09-19:
Thanks a lot for your reply,
* my question may sound dump to an expert like you, but it was general, like, why don't we do the POST request rather than doing the GET request? is that the case with most of the APIs ? and why is that ?
* I know in HTTP we can sniff the packets with wireshark for example "if we're on LAN" and they will show up in plain text "even the body of the request where we can see the POST content" .. 
but what about encoding? what does the hacker need to decode the URI then ? in encryption we need a key, what about it in encoding ?

Thanks a lot, and your reply is much appreciated ...

tttapa, 2019-09-19:
GET is used to retrieve data from the server, POST is used to submit data to the server. There are some major differences, such as idempotence, safety (of the method, not data safety, see [RFC7231 §4.2.1](https://tools.ietf.org/html/rfc7231#section-4.2.1)), cachability, etc.  
That being said, not all APIs follow this convention, and GET requests are sometimes easier to parse or compose.

Encoding and encryption are completely different.  
Encoding is just changing the representation of the data. For example, in URI encoding, reserved characters are replaced by a percent sign followed by their UTF-8 representation as hexadecimal numbers. For example, a space is replaced by `%20`.   
There's nothing secret about encoding, anyone can convert it back to its original representation without requiring a secret keys.

Think of UTF-8 vs UTF-16 encoding. The meaning of a piece of text in either encoding is the same, but the bytes that are saved to your hard drive are different.

Mohammad271091, 2019-09-20:
You're awesome man, I have a good idea about web development "especially PHP" and about security "like TLS, hashing, encryption,etc..."
But I just got confused when it came to ESP8266... 
I'm very grateful and I don't wanna waste your time with in-depth stuff not exactly related to ESP8266.

What I want to know now, if a hacker was on the same network with the esp8266 gmail notifier, with a wireshark, if it also was a HTTP connection of the ESP, he can get the credentials and decode them easily, no? "Supposed that the hacker knows the charset used".. if it was a HTTPs he will then need the encryption key "or called TLS key, not sure" to decrypt the traffic then decode ... 
This applies in both GET and POST requests.
Is that correct what I said? I hope that I'm having a clue with what I just said.. and I hope you correct me if I'm wrong.. 
I love you man! You're really awesome..

tttapa, 2019-09-20:
If the email notifier used HTTP, then anyone that happened to be on a network that your packet was sent over could read your email address and password. It doesn't have to be your home network, for example. And this would be true for all HTTP methods, POST is not safer than GET.

Luckily, the example uses the WiFiClientSecure class, which means it uses TLS, and a hacker would need a key to decrypt it. That still doesn't mean that your credentials are safe though, because an attacker could pretend to be the Google server, and then you send your credentials to a fake server. It doesn't matter if the connection itself is encrypted using TLS, your data still ends up at the hacker's computer.  
To stop this from happening, certificates exist: You check the server's fingerprint against the fingerprint on the certificate, if it matches, you can be sure that it's the real server. That's what the line `client.verify(fingerprint, host)` does in the example.

Mohammad271091, 2019-09-21:
You're making me satisfied with your answers ! Thanks!! 
I think you meant the MitM attack with the example you mentioned.. 
Whats great about working with esp8266 is that it lets me understand and manually do the things we used to have it already done in the background without even knowing about it! "E.g: verifying the certificate and creating a TLS instance" .. you have really done a great favor with your tutorial. And if it didn't exist I guess I'd never learn that much about it and interlink all things together.. 
One last question, how does the esp generate an encryption key? Is this process done in the wificlientsecure class? What can we do to prevent the mitm attack with esp? .. I'm asking this in general and not example specific .. to learn from you .. 
I'd like to know your name as well :) .. my pleasure .. and happy to talk to gentle man like you :) 

Thanks in advance

tttapa, 2019-09-25:
Yes, I was talking about a man in the middle attack.

The encryption is handled by other libraries in the ESP core. The Arduino core for ESP8266 is open-source, so you can do some digging if you want.

Mohammad271091, 2019-09-28:
one LAST question.. 
I sent a post request from nodemcu to a php on 'localhost running apache "MAMP"' as follows:
$x = "ok";
$postData = "station=" + x ; // this should send the key=value as station = ok
then sent the request with http.POST with a content-type header of"application/x-www-form-urlencoded".. I also included to see the response return code and payload.. everything was ok.. 

on the PHP handler I wrote:
$data = $_POST['station'];
echo $data;

The issue is that I saw the "data = ok"  and the code "200" on the serial monitor which is great..
but why I can't see echo result "data = ok" on the browser of my localhost ?? I believe that in this case the client (browser role) is the nodemcu and not the browser on the compute running the server for sure! .. but since I can use the posted data -as a server- like to insert them in a database, and processing with if statements for example and so on... why I can't echo them directly ?! and what if I want to ? it prints everything except the posted values, only can be shown on the serial monitor !.. 
I know Im running around the answer but Im confused ... this will be a last question .. sorry for asking too much...

Thanks ..

tttapa, 2019-09-28:
I don't understand your setup. Why do you respond with the data you just posted? If the ESP sent the post data, then why does the server send it back?

How are you sending post requests using the browser? $_POST only contains the data of the current request. So if you post the data using the ESP, and then open the same url in a browser, the server has long forgotten the post data. Each PHP script is run once for each request, and they are completely separate.

Mohammad271091, 2019-09-28:
told you, confused :( ..
what I did is putting the same content in the POST request just for testing.. it was "data = ok" and sending it every 2 seconds "included a delay after each request"..
what confused me is that why the echo command showed the output on the Serial monitor and not on the browser running the PHP script!... and what to do if I wanna do that ? I want to echo the "data = ok" on the localhost browser whenever its sent by the nodemcu ...