tttapa/ESP8266 #67: Ct arm computability

ShelbyTech95, 2019-10-06:
Can i know if it is compitable for ct arm to connect with esp8266?

tttapa, 2019-10-07:
I don't know what a CT ARM is, but if the second page of Google is to be trusted, it looks like a Cortex M0+ microcontroller. If that's the case, then yes, you can connect it to an ESP8266.

ShelbyTech95, 2019-10-13:
Okay sir, good to know that. Thank you for the reply. One more question sir. Can you suggest any IoT platform that do not need the type for microcontroller because in blynk apps there is no option for ct arm microcontroller.

tttapa, 2019-10-13:
I'm sorry, I don't know anything about the microcontroller you mention, and I'm not really up-to-date with IoT right now, so I'm afraid I can't really help you.