tttapa/ESP8266 #72: SPIFF - no function under "Tools" to upload data to SPIFF - plugin needed?

k2OS, 2019-12-29:
I am following your great tutorial.. but under SPIFF there seems to be something amiss.. when instructing how to upload the data-folder to SPIFF, you mention a feature that only seems to be accesible with a plugin? I can't find any "ESP8266 sketch data upload" menu point anywhere.. 
A google search leads me to a plugin from 2015 or 2016 to add this feature to the IDE..?

ilioss, 2020-06-17:
Hi,
Have a look at random nerts webpage for SPIFFStutorial.
Succes.
IlioSS

Anton-V-K, 2020-10-21:
The plugins like https://github.com/esp8266/arduino-esp8266fs-plugin and https://github.com/earlephilhower/arduino-esp8266littlefs-plugin still work fine with Arduino IDE 1.8.13 (on Windows)