tttapa/ESP8266 #76: Regarding methods of coding esp8266

gkunalupta, 2020-05-05:
What is role of At commands n esp8266.

The function which we use in arduino ide for getting connecting to wifi and server do the use At commands or they use some sort of registers or their function definition has something to do with TCP/IP stack

tttapa, 2020-05-06:
AT commands were used in the 1980s to talk to modems over a serial port. If you install AT firmware to the ESP8266, you can also use these commands to talk to the ESP, over its serial port.

IMHO, there are very few scenarios where this is actually a good idea. It's often used to communicate between an Arduino UNO and an ESP8266, which is just ridiculous if you think about it: the ESP8266 is many times faster than the UNO and has many times the memory, yet the ESP is used as a dumb slave, while the UNO handles all heavy lifting like parsing requests, serving web pages, etc. All of the network traffic has to go over the slow serial link, including all the parts you don't care about.

A much better idea would be to program the ESP8266 directly, so you can fully leverage all of its processing power and available memory (both flash and RAM).  
Then you only send minimal data over the serial link between the ESP and the UNO, or you leave out the UNO entirely, and handle everything on the ESP.

gkunalupta, 2020-06-06:
okay and thank you for that suggestion and sry for late reply.
And other thing for Using Arduino uno .... I have done that thing and that works perfectly using hardware serial library here my concern is to learn the.embedded c programming and sharp my skills for firmware programming of different pheriferals like UART ... developing the whole libraries for by own for esp8266 and 32 with AT commands  will teach me many new things to know c and embedded c language and what i think that's how i can develop my skills for firmware programming .