tttapa/ESP8266 #81: How can i use ESP8266 as a WI-FI adaptor?

the-hk, 2020-07-21:
i mean is there any code or way to connect my computer which has not wifi module to connect my modem and i have esp8266 and i would like to use this module as a wifi module is it possible?

tttapa, 2020-07-21:
I'm sure it's possible, but it's probably not worth the time investment. Especially since USB WiFi dongles are so cheap.

aramperez, 2020-07-22:
Someone correct me, but I don't believe it is possible. When the ESP8266 is in AP mode, it can NOT connect to the Internet.

tttapa, 2020-07-22:
> Someone correct me, but I don't believe it is possible. When the ESP8266 is in AP mode, it can NOT connect to the Internet.

Why would the ESP8266 need to be in AP mode?  
Also, you can use the WiFi in AP and station mode at the same time: https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html?highlight=WIFI_AP_STA#mode

aramperez, 2020-07-22:
Yes, but I would not recommend it, search for "esp8266 sta ap channel issues".

juancb, 2023-02-17:
What about using it as a wifi interface for another microcontroller board like an Arduino UNO R3?