tttapa/ESP8266 #84: mDNS not working

jtannenba, 2020-11-16:
This is driving me crazy.  Did all according to the instructions but it seems that xxxxx.local doesn't work on my system.  I have CenturyLink and an Actiontec C1900A router.  Looked at all the settings for the router, but could not find any pertinent setting.
this doesn't work from either my Windows 10 machine or linux machine which are on ethernet cable to an exp8266 which is on wifi.
It does work however between wired hosts.  Am I missing a setting on my router?

yomas000, 2020-12-02:
It worked for me but you have to do MDNS.update(); in the void loop.

bass1959, 2021-03-06:
Yeah, that's exactly what I was about to mention..
Would be great if the author added this to his terrific guide.